thortex/rpi3-webiopi #12: my solution for RPI3 WebIOPi - ERROR - global name 'GPIO' is not defined fixed

thortex, 2016-09-12:
https://groups.google.com/forum/#!topic/webiopi/BGc-mcD3UEw

logiccore AU 	
8月16日
メッセージを次の言語に翻訳: 日本語  
Hello I´m from colombia, i have been try to fix this problem 2 weeks ago. 

i´m using a RPI3 B+   OS: jessie litte

Here the problem after running Webiopi default example 

  File "/usr/lib/python3.4/socketserver.py", line 305, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.4/socketserver.py", line 331, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python3.4/socketserver.py", line 344, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python3.4/socketserver.py", line 669, in __init__
    self.handle()
  File "/usr/lib/python3.4/http/server.py", line 398, in handle
    self.handle_one_request()
  File "/usr/lib/python3.4/http/server.py", line 386, in handle_one_request
    method()
  File "/usr/local/lib/python3.4/dist-packages/WebIOPi-0.7.1-py3.4-linux-armv7l.egg/webiopi/protocols/http.py", line 267, in do_GET
    self.processRequest()
  File "/usr/local/lib/python3.4/dist-packages/WebIOPi-0.7.1-py3.4-linux-armv7l.egg/webiopi/protocols/http.py", line 258, in processRequest
    except (GPIO.InvalidDirectionException, GPIO.InvalidChannelException, GPIO.SetupException) as e:
NameError: name 'GPIO' is not defined

How i fixed the error

1. run cat /proc/cpuinfo

will get revision number as a22082

2. then add this number in the cpuinfo.c located in /home/pi/WebIOPi-0.7.1/python/native
as  i show here in the next code. hope it work for you 

int get_rpi_revision(void)
{
   char revision[1024] = {'\0'};
   
   if (get_cpuinfo_revision(revision) == NULL)
      return -1;
      
   if ((strcmp(revision, "0002") == 0) ||
       (strcmp(revision, "1000002") == 0 ) ||
       (strcmp(revision, "0003") == 0) ||
       (strcmp(revision, "1000003") == 0 ))
      return 1;
   else if ((strcmp(revision, "0004") == 0) ||
            (strcmp(revision, "1000004") == 0 ) ||
            (strcmp(revision, "0005") == 0) ||
            (strcmp(revision, "1000005") == 0 ) ||
            (strcmp(revision, "0006") == 0) ||
            (strcmp(revision, "1000006") == 0 ))
      return 2;
   else if ((strcmp(revision, "a01041") == 0) ||
            (strcmp(revision, "a21041") == 0 )||
            (strcmp(revision, "a22082") == 0 ))
      return 3;
   else // assume rev 4
      return 4;

closed 2016-09-12 by thortex

Deepfriedrolos, 2018-06-02:
I have a Raspberry Pi3, i tried your code but then webiopi doesnt work. Is there anywhere i can get the code in a file format as copy and paste looses the indents. Im not really a coder, sorry.

Deepfriedrolos, 2018-06-02:
Hi,

Webiopi seems to be a dead project now. Could you help me fix this error on my Pi3? 

I cannot copy/paste your code and get it to work unfortunately, it breaks the webiopi setup.sh process.

Thanks

commit 1ecf3ef5 (2016-09-12): close #12