thortex/rpi3-webiopi #17: "systemctl daemon-reload" is recommended in jessie

thortex, 2016-09-15:
If /etc/systemd/system/webiopi.service file is updated by dpkg, daemon reload is recommended in jessie:

    sudo systemctl daemno-reload webiopi

"postinst" file should include above sequence to reload daemon automatically.

nealre, 2017-02-03:
All installed very well had no problems. It asked for a password on the Web when i went to look at the Webiopi mapping and  nothing i did worked?
Help
Rick
nealre990@gmail.com

thortex, 2017-02-03:
Username is webiopi, and password is raspberry as default.

nealre, 2017-02-04:
thank you, that was what I needed. After logging in to the GPIO-header does not display? Just a blank screen? Neither does GPIO-list. When you select either it moves to the display screen but it is blank? If i try to run webiopi it says the ip is in use.
Not sure what to do at this piont as i don't have an understanding how the software flows.
Rick

thortex, 2017-02-05:
could you check the following points to know what is wrong? 

1. Did you install 'rpi3-webiopi' raspbian package? Both rpi2-webiopi and rpi3-webiopi package is required to use WebIOPi under python3.
2. Did you install the official WebIOPi before installing Yet Another WebIOPi? 
3. Do you use another process which is using socket port number 8000 (e.g., squid, Apache tomcat, et. cetera)? 
4. Could you execute 'webiopi-diag' command, and send log message to analyze the issue?

nealre, 2017-02-06:
Thortex
I did not know there were two wepiopi types (RPi2, RPi3). So i'm not sure which I loaded. I ran the diag file and it was very interesting but I don't know how to forward it to you through the RPi or if it's in winSCP i can't find it? Sorry but my learning curve is still climbing. As the port goes , No nothing else it running on that port. I will try to find where to download the webiopi (PI2,Pi3) from?
Rick

thortex, 2017-02-07:
sorry, a typo. 

either python2-webiopi, or python3-webiopi instead of rpi2-/rpi3-webiopi.

nealre, 2017-02-07:
Thortex here is the diag you ask me to pull.   
[webiopi-diag-log-20170205-233011.txt](https://github.com/thortex/rpi3-webiopi/files/756687/webiopi-diag-log-20170205-233011.txt)
Thanks for you help
Rick

thortex, 2017-02-07:
The following instruction may be helpful to resolve the issue.

Primary, Remove dpkg packages.

    $ sudo dpkg --purge python3-webiopi 
    $ sudo dpkg --purge python2-webiopi 

Secondary, Create an old file list.

    $ find /usr | grep -i webiopi  | sort > ~/webiopi.old.files

Third, Create an old file archive. Your original files are backed up in the ~/webiopi.old.tbz

    $ tar cjvf ~/webiopi.old.tbz --files-from ~/webiopi.old.files 

Forth, Remove old files completely.

    $ cat ~/webiopi.old.files | sort -r | sed -e 's/^/sudo rm -f /;' |  sh 2>/dev/null
    $ cat ~/webiopi.old.files | sort -r | sed -e 's/^/sudo rmdir /;' | sh 2>/dev/null

Fifth, Install dpkg packages.

    $ sudo dpkg -i ~/build.webiopi/python2-webiopi_0.7.1*.deb
    $ sudo dpkg -i ~/build.webiopi/python3-webiopi_0.7.1*.deb

Sixth, Select your preferred python version for YA-WebIOPI.

    $ sudo webiopi-select-python 2
or 

    $ sudo webiopi-select-python 3

Finally, 

     sudo systemctl daemon-reload
     sudo systemctl restart webiopi 

or 

    $ sudo /etc/init.d/webiopi restart

nealre, 2017-02-08:
thortex,
can't thank you enough as this thing, like several other has experienced, has driven me crazy, but it works now. I was researching and found in rest.py where it names GPIO but wasn't sure what to do.
Thanks 
Rick

nealre, 2017-02-08:
AT 3am this  morning I saw the in/out for the first time so i went to bed. Today as i set it up again and was going to go through it to test pins you can't set them. Unless i'm  missing something. I thought all you had to do was click on a pin to change the state? I tried and none of them respond. My project is wired to GPIO 17/18 two leds just to verify functionality before moving on.
Please advise
Rick

nealre, 2017-02-08:
Sorry, they don't change states, nothing changes in/out can't be changed. also, the device GPIO screen does not come up. The header and monitor, and list screens do.
Hope that helps
Rick

nealre, 2017-02-09:
Thorex
I formated a new SD card and downloaded NOOBE, installed it and loaded Python2 and 3. the webiopi and after that  i don't remember all of it but as crazy as it sounds it worked. I tried to do a webiopi-diag but it would not find the file to run it? I was going to send it to you so you could learn too but  the diag must be looking in webiopi and or something?
any way i'll try again later. 
Thought it may make you feel good...
Rick