thortex/rpi3-webiopi #27: PWM output does not work

RemiSlv, 2016-09-19:
Hi,

I am having issues with PWM controls on the REST API:

`POST /GPIO/x/function/pwm returns` a 403

`POST /GPIO/x/pulseRatio/0.5` results in a single one period of PWM

`POST /GPIO/x/pulseRatio/1` has the same effect as `POST /GPIO/x/value/1`

Config: RPi3 / Python 2.7.9 / rpi3-webiopi-0.7.1-deb03

The same issue was reported earlier on https://groups.google.com/forum/#!topic/webiopi/qN0RWDxPPO4

thortex, 2016-09-20:
I was able to reproduce the problem.
REST interface for pwm is not provided in v0.7.1.
We should add the interface, if you would like to pwm function for GPIO.

RemiSlv, 2016-09-21:
Hi Watanabe-san. Thank you so much for picking this up. This will be very useful to me if you find the way to fix it ! good luck

thortex, 2016-09-21:
I've fixed the issue, and committed to master branch (as a work-around):
  https://github.com/thortex/rpi3-webiopi/commit/d6ea7f688c62197fdfbfc95661dc9c4b2ea8424f

Test script is also committed:
https://github.com/thortex/rpi3-webiopi/blob/master/dev/test/hotaru.sh
Please modify user, pass, ipadr, port, and pin parameter (line 4 through 8) for your RPi environment (e.g. An LED is wired to my RPi GPIO port no. 4).
https://github.com/thortex/rpi3-webiopi/blob/master/dev/test/hotaru.sh#L8

pulseRatio and pulesAngle is already deprecated in v0.7.1.
But the following REST interfaces are not provided yet:
  GET /GPIO/%d/pwm
  POST /GPIO/%d/pwm/enable
  POST /GPIO/%d/pwm/disable
  POST /GPIO/%d/pwmWrite
  POST /GPIO/%d/pwmWriteAngle
Above interfaces may be added in the future release.

RemiSlv, 2016-09-22:
That is great ! Confirmed it is working accordingly from my end as well.
As mentioned in your test script, PWM default frequency is 50Hz. 

Is there a method to change it ? I can't figure this out easily from the code. I'm using it to control LEDs so I'd like to change that to 100Hz or so to reduce the flickering effect :)

Other than that, this issue can be closed !

Thanks :+1:

thortex, 2016-09-23:
I added pulse frequency setting interface to REST.
Please test with /dev/test/ directory scripts: 
  https://github.com/thortex/rpi3-webiopi/blob/master/dev/test/

maximum frequency is 100.0 Hz, but 100.0 Hz is not recommended for ordinary use.

    ./hotaru.sh 100

In case of 50 Hz: 

    ./hotaru.sh 50

In case of three seconds blinking at 1 Hz:

    ./min-pulse.sh 1 3

In case of three seconds blinking at 2 Hz: 

    ./min-pulse.sh 2 3 

In case of two seconds blinking at 10 Hz: 

    ./min-pulse.sh 10 2

In case of twice sos signals for emergency (sleep timing should be optimized for your CPU frequency):  

    ./sos.sh

Usage restriction is listed below:

* First, set GPIO pin function to PWM, then set pulse frequency, Finally set pulse ratio.
* For blinking, pulse frequency can be set to less than 25 Hz.
* For dimmer,  pulse frequency can be set to more than 30 Hz.
* Does not work at less than 1 Hz.

RemiSlv, 2016-09-23:
This is excellent, thank you so much again. 
I'm all set now, you've been a great help !

Just out of curiosity, may I know why 100Hz is not a recommended value ?

thortex, 2016-09-23:
Please refer to WebIOPi forum in Google Groups:
https://groups.google.com/forum/#!topic/webiopi/qN0RWDxPPO4

RemiSlv, 2016-09-23:
thanks, I am closing this issue now

closed 2016-09-23 by RemiSlv

thortex, 2016-09-23:
Thank you for your issue reporting, and closing.
Please feel free to ask me, if yo have any questions, ideas, or improvements.