thortex/rpi3-webiopi #32: Error while trying to install

steven6282, 2016-10-27:
I was following the installation guide and made it to this step:
./10_make_deb.sh
This errors on my rpi3 with:
`native/bridge.c: In function ‘py_set_upllupdn’:
native/bridge.c:203:2: warning: implicit declaration of function ‘set_pullupdn’ [-Wimplicit-function-declaration]
  set_pullupdn(channel, pud);
  ^
native/bridge.c:184:6: warning: unused variable ‘function’ [-Wunused-variable]
  int function;
      ^
native/bridge.c: At top level:
native/bridge.c:1244:31: error: ‘py_set_pullupdn’ undeclared here (not in a function)
  {"setPullUpDn", (PyCFunction)py_set_pullupdn, METH_VARARGS | METH_KEYWORDS, "Setup the GPIO pull/up down control\nchannel   - BCM GPIO number\n\npull_up_down - PUD_OFF, PUD_UP or PUD_DOWN"},
                               ^
native/bridge.c:177:18: warning: ‘py_set_upllupdn’ defined but not used [-Wunused-function]
 static PyObject *py_set_upllupdn(PyObject *self, PyObject *args, PyObject *kwargs)
                  ^
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
dh_auto_build: python setup.py build --force returned exit code 1
debian/rules:7: recipe for target 'build' failed
make: *** [build] Error 1
dpkg-buildpackage: error: debian/rules build gave error exit status 2
debuild: fatal error at line 1376:
dpkg-buildpackage -rfakeroot -D -us -uc failed`

thortex, 2016-10-29:
Thank you for your bug reporting and providing the solution.
I fixed the typo.

thortex, 2016-11-01:
Hi, steven6282.

I've committed modified source code for the issue #32 with c34cf0f.
Can I close this issue?

closed 2016-11-13 by thortex