thortex/rpi3-webiopi #33: GPIO Header and List Blank

steven6282, 2016-10-27:
I just submitted an error a little bit ago about an error while installing which I fixed by changing a assumed typo.  But I'm having another problem post install and don't know what is causing it.

When I run the webiopi with webiopi -d -c /etc/webiopi/config, and then go to the server:8000, it loads the web page.  But then when I go to GPIO Header, or GPIO List the pages are completely blank.

I found this fork of webiopi when I was researching a problem with the original webiopi not working on my rpi3.  I found a thread talking about the cpu revision so I don't know if that has something to do with it or not.  I don't see revisions specified in cpuinfo.c like they did and modified to fix their problem.  But just in case it matters cat /proc/cpuinfo shows my revision as a02082

thortex, 2016-10-29:
Raspberry Pi 3 Model B (Q1 2016) rev. a0282 is supported on YA-WebIOPi+.

Root privilege is not needed for the official WebIOPi, but root privilege is required to execute Yet Another WebIOPi+. So, Could you execute YA-WebIOPi+ with sudo?
  $ sudo webiopi -d -c /etc/webiopi/config -l /var/log/webiopi

Or, did you run dpkg command to install YA-WebIOPi+?
Please refer to https://github.com/thortex/rpi3-webiopi/wiki/HowToBuild for installation information.

Guru-98, 2016-10-30:
Same error here...

A little debugging showed me an error at webiopi.js

Uncaught TypeError: Cannot set property 'mapped' of undefined @ webiopi,js:222

thortex, 2016-10-30:
Could you send me console logs by typing the following commands?
 $ ps aux | grep webiopi | grep -v grep
 $ egrep -v ' (DEBUG|INFO) ' /var/log/webiopi | tail -100
 $ lsb_release -a
 $ cat /proc/cpuinfo | egrep '(Hardware|Revision)'
 $ find /usr/share/webiopi/htdocs/ | wc -l
 $ uname -a
 $ grep -v '^#' /etc/webiopi/config | grep -v '^$'
 $ ls -lh `which python`
 $ dpkg -l | egrep '(webiopi|setuptools|on-dev|on3-dev|gpio)'
 $ sudo webiopi -d -l /var/log/webiopi -c /etc/webiopi/config
 $ curl -u webiopi:raspberry http://localhost:8000/
 $ curl -u webiopi:raspberry http://localhost:8000/app/gpio-header
 $ curl -u webiopi:raspberry http://localhost:8000/map
 $ curl -u webiopi:raspberry http://localhost:8000/version
 $ curl -u webiopi:raspberry 'http://localhost:8000/GPIO/*'

thortex, 2016-10-30:
I've added syslog debugging log outputs to native C source files:
  https://github.com/thortex/rpi3-webiopi/commit/98c847dae5dc7b695ffa4293a8d093d196f66fc1

If you can use the above latest master branch, in addition, could you execute the following commands finally?
 $ sudo grep python /var/log/syslog

Guru-98, 2016-10-30:
Open the [log](https://github.com/thortex/rpi3-webiopi/files/560408/log.txt) using cat command... if you use gedit its going to be a bit crappy :grimacing: 

### Log from the new commit

```shell
Oct 30 12:32:24 devpi1 python3[364]: 2016-10-30 12:32:24 - WebIOPi - INFO - Starting WebIOPi/0.7.1/Python3.4
Oct 30 12:32:24 devpi1 python3[364]: 2016-10-30 12:32:24 - WebIOPi - INFO - GPIO - Native mapped to REST API /GPIO
Oct 30 12:32:24 devpi1 python3[364]: 2016-10-30 12:32:24 - WebIOPi - INFO - Loading configuration from /etc/webiopi/config
Oct 30 12:32:24 devpi1 python3[364]: 2016-10-30 12:32:24 - WebIOPi - INFO - Access protected using /etc/webiopi/passwd
Oct 30 12:32:24 devpi1 python3[364]: 2016-10-30 12:32:24 - WebIOPi - INFO - HTTP Server binded on http://localhost:8000/
Oct 30 12:32:24 devpi1 python3[364]: 2016-10-30 12:32:24 - WebIOPi - INFO - CoAP Server binded on coap://localhost:5683/
Oct 30 12:32:24 devpi1 python3[364]: 2016-10-30 12:32:24 - WebIOPi - ERROR - [Errno 19] No such device
Oct 30 12:32:24 devpi1 python3[364]: Traceback (most recent call last):
Oct 30 12:32:24 devpi1 python3[364]: File "/usr/local/lib/python3.4/dist-packages/WebIOPi-0.7.1-py3.4-linux-armv7l.egg/webiopi/__main__.py", line 75, in <module>
Oct 30 12:32:24 devpi1 python3[364]: main(sys.argv)
Oct 30 12:32:24 devpi1 python3[364]: File "/usr/local/lib/python3.4/dist-packages/WebIOPi-0.7.1-py3.4-linux-armv7l.egg/webiopi/__main__.py", line 69, in main
Oct 30 12:32:24 devpi1 python3[364]: server = Server(port=port, configfile=configfile, scriptfile=scriptfile)
Oct 30 12:32:24 devpi1 python3[364]: File "/usr/local/lib/python3.4/dist-packages/WebIOPi-0.7.1-py3.4-linux-armv7l.egg/webiopi/server/__init__.py", line 130, in __init__
Oct 30 12:32:24 devpi1 python3[364]: self.coap_server.enableMulticast()
Oct 30 12:32:24 devpi1 python3[364]: File "/usr/local/lib/python3.4/dist-packages/WebIOPi-0.7.1-py3.4-linux-armv7l.egg/webiopi/protocols/coap.py", line 477, in enableMulticast
Oct 30 12:32:24 devpi1 python3[364]: self.socket.setsockopt(socket.IPPROTO_IP, socket.IP_ADD_MEMBERSHIP, mreq)
Oct 30 12:32:24 devpi1 python3[364]: OSError: [Errno 19] No such device
Oct 30 16:28:33 devpi1 python3[23230]: 2016-10-30 16:28:33 - WebIOPi - INFO - Starting WebIOPi/0.7.1/Python3.4
Oct 30 16:28:33 devpi1 python3[23230]: 2016-10-30 16:28:33 - WebIOPi - INFO - GPIO - Native mapped to REST API /GPIO
Oct 30 16:28:33 devpi1 python3[23230]: 2016-10-30 16:28:33 - WebIOPi - INFO - Loading configuration from /etc/webiopi/config
Oct 30 16:28:33 devpi1 python3[23230]: 2016-10-30 16:28:33 - WebIOPi - INFO - Access protected using /etc/webiopi/passwd
Oct 30 16:28:33 devpi1 python3[23230]: 2016-10-30 16:28:33 - WebIOPi - INFO - HTTP Server binded on http://172.17.21.205:8000/
Oct 30 16:28:33 devpi1 python3[23230]: 2016-10-30 16:28:33 - WebIOPi - INFO - CoAP Server binded on coap://172.17.21.205:5683/
Oct 30 16:28:33 devpi1 python3[23230]: 2016-10-30 16:28:33 - WebIOPi - INFO - CoAP Server binded on coap://224.0.1.123:5683/ (MULTICAST)
```

thortex, 2016-10-30:
hmm.

1. Kill all WebIOPi daemons before starting a new process instance by using systemctl, /etc/init.d/webiopi, or kill command. Yet Another WebIOPi+ Debian package automatically installs systemd script for jessie, or /etc/init.d script to daemonize WebIOPi process.
2. change your WebIOPi HTTP port number in /etc/webiopi/config, if another process is already using tcp port 8000.
3. Downgrade/Upgrade your Kernel by rpi-update.
4. Disabling COAP by editing /etc/webiopi/config.
5. Change your web browser instead of a preferred one, WebIOPi seems to be working normally.

Guru-98, 2016-10-31:
Done every thing you said... The problem persists

Getting the revision from the server gives me the wrong revision

```shell
$curl -u webiopi:raspberry http://localhost:8000/revision
2
```

and so wrong mapping too...

```shell
$curl -u webiopi:raspberry http://localhost:8000/map
2
```

but running the python file `version.py` in `webiopi.utils` separately gave me `BOARD_REVISON` as 3... :confused: :roll_eyes:

thortex, 2016-10-31:
Jesus!

I realized that.
Your information was useful for me, thanks a lot.

I committed the modified python code to: 
https://github.com/thortex/rpi3-webiopi/commit/ab221ff118ef6cb838f1212d8ba5e41978601fd7

I've tested with RPi1(BCM2708/000d) and RPi3(BCM2709/a02082).

Please check ab221ff out.

Guru-98, 2016-10-31:
Happy to help....

thortex, 2016-11-01:
Hi, steven6282 .

I've committed modified source code for the issue #33 into ab221ff.

In case of Guru's RPi board, the problem was solved. How about you?

Guru-98, 2016-11-02:
I hate to say this, but the problem still persists... :disappointed: I used your `syslog` function to view the `BOARD_REVISION` directly from the native gpio library `gpio.c` but when `curl`-ing the module returns revision as 2... :confused:

thortex, 2016-11-04:
I made a diagnostic command  and committed to. Could you send a result of the following commands:
    curl https://raw.githubusercontent.com/thortex/rpi3-webiopi/master/webiopi_0.7.1/python/native/webiopi-diag > webiopi-diag
    chmod +x webiopi-diag
    ./webiopi-diag

thortex, 2016-11-05:
Sorry, I renamed the file name: webiopi-diag to webiopi-info;
https://github.com/thortex/rpi3-webiopi/blob/master/webiopi_0.7.1/python/native/webiopi-info

jcarty, 2017-01-13:
Will this be fixed soon? @thortex Seeing the same

thortex, 2017-01-13:
In case of my environments (both RPi1b and RPi3b), I can't face the problem. 

Furyher information will be useful for me (e.g., screenshots, the kernel version, the board revision, official WebIOPi existence, w/ or w/o PIXEL, Python module installation status, debug mode logs, wireshark/tcpdump packet capture logs, et  al.).

thortex, 2017-01-15:
> Note that remove either the official WebIOPi or the yet another WebIOPi python codes/modules/eggs from your system before installing the latest Yet Another WebIOPi+ debian package. You can install either the official WebIOPi or YA-WebIOPi+.

> python3-webiopi package only includes a native c library and python module scripts without htdocs, init, systemd, and etc. Even if you only use python3, you have to install python2-webiopi debian package to work proper.

nealre, 2017-02-07:
Thortex
 I tried your rpi3-webiopi "how to build your own..... and now the gpio display comes up but we are back to the pins  not being numbered, both gpio screens.
Rick

PatSherbrooke, 2017-07-11:
Does anyone has the blank page on <IP:8000> on the **Ubuntu Mate 16.04.2 environment** ?
It's my case and I don't know how to fix it...
I'm stucked in my project due to this error.
I cannot switch to Raspbian for other reasons...
If someone have some clues...it would be great !!!
Thanks in advance for your answers...