thortex/rpi3-webiopi #41: PWM functions don't work on newer kernel version.

thortex, 2016-12-09:
There meight be  modification(s) about hardware PWMs (or analog audio) on newer Raspbian kernel.
PWM functions don't work on newer kernel version with RPi3.

dtoverlay=pwm-2chan,pin=12,pin2=13,func=4,func2=4 is needed on /boot/config.txt? or pwm-bcm2835.ko is needed?

bill-orange, 2017-03-17:
Was this ever addressed?  I am also having problems with wiringpi on the new kernal

thortex, 2017-03-17:
if you use hardware PWM feature, you should read /boot/overlays/README: 
  https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README#L991

If you need not use onboard analog audio, you can use hardware PWM feature.
  https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README#L1000

As a work-around, add the following lines in /boot/config.txt to use single channel hardware PWM:

     dtoverlay=pwm,pin=18

, or to use two channels: 

     dtoverlay=pwm-2chan,pin=18,pin2=19

And, remove the following line, if it exists:

     dtparam=audio=on

thortex, 2017-03-22:
Do you have any troubles on wiringPi or wiringPi-Pytyon yet?
If you use PWM feature on wiringPi, you have to run python command with sudo.

Hardware PWM feature is supported on git repository latest version, the setup instruction is listed below: 

1. sudo apt-get update && sudo apt-get upgrade -y
2. sudo apt-get install -y python-dev python-setuptools swig python3-dev python3-setuptools
3. git clone https://github.com/Gadgetoid/WiringPi2-Python.git
4. cd WiringPi2-Python
5. wget -O wiring.tar.gz 'https://git.drogon.net/?p=wiringPi;a=snapshot;h=96344ff7125182989f98d3be8d111952a8f74e15;sf=tgz'
6. tar xzf *.gz
7. cp -r wiringPi*/* WiringPi
8. perl -i.bak -pe "s/('_wiringpi2',)/\$1\n    extra_link_args=['-lrt','-lcrypt'],/;" WiringPi*/setup.py
9. rm wiringPi*/devLib/piFaceOld.c
10. perl -i.bak -pe 's/PyString_FromStringAndSize/PyUnicode_FromStringAndSize/;' ./WiringPi2*/wiringpi.i
11. cd WiringPi
12. swig2.0 -python -threads wiringpi.i
13. sudo python setup.py build install
14. python3 setup.py build install
15. python test.py
16. python3 test.py

bill-orange, 2017-03-24:
I abandoned WiringPI and used RPi.GPIO instead.  RPi.GPIO works fine and I don't see a down side to the switch.