thortex/rpi3-webiopi #43: CoAP Python client returns "No data received" exception

sumghai, 2016-12-24:
I'm currently developing an application where one RPi 3 tells a second RPi 3 over a local wireless network to run selected macros (`conditionCruise`, `conditionYellow` and `conditionRed`):

````
from webiopi.clients import *
import webiopi
import time

# Connect to the RPi 3B+ running the Bridge Alert Light script
# Credentials are not needed, since both will be on a private network
client = PiHttpClient("192.168.1.101") 

# Setup function is automatically called at WebIOPi startup
def setup():	
	webiopi.sleep(1)

# Loop function is repeatedly called by WebIOPi 
def loop():
	value = Macro(client, "conditionCruise").call()
	time.sleep(6)
	value = Macro(client, "conditionYellow").call()
	time.sleep(6)
	value = Macro(client, "conditionRed").call()
	time.sleep(6)
	webiopi.sleep(0.05)
	
# Destroy function is called at WebIOPi shutdown
def destroy():
	webiopi.sleep(1)
````

The HTTP Python client works fine, but I'd like to switch to the CoAP Python client as a prelude to multicast communications to multiple remote RPis.

However, when I changed `PiHttpClient` to `PiCoapClient`, and enabled CoAP on my first Pi, the macros on the remote Pi aren't triggered, and the CoAP server hangs. 

Running the WebIOPi service on the first Pi in debug mode reveals the following error message: 

````
2016-12-25 12:02:53 - WebIOPi - INFO - Loading hsfm-entd-bridge-ctrlnode from /home/pi/hsfm-entd-bridge-ctrlnode/python/pi2pi_test.py
2016-12-25 12:02:54 - WebIOPi - INFO - Passwd file /etc/webiopi/passwd is empty
2016-12-25 12:02:54 - WebIOPi - WARNING - Access unprotected
2016-12-25 12:02:54 - WebIOPi - INFO - HTTP Server binded on http://192.168.1.100:8000/
2016-12-25 12:02:54 - WebIOPi - INFO - CoAP Server binded on coap://192.168.1.100:5683/
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.4/threading.py", line 920, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.4/dist-packages/WebIOPi-0.7.1-py3.4-linux-armv7l.egg/webiopi/utils/thread.py", line 23, in run
    self.func()
  File "/home/pi/hsfm-entd-bridge-ctrlnode/python/pi2pi_test.py", line 19, in loop
    value = Macro(client, "conditionCruise").call()
  File "/usr/local/lib/python3.4/dist-packages/WebIOPi-0.7.1-py3.4-linux-armv7l.egg/webiopi/clients/__init__.py", line 107, in call
    return self.sendRequest("POST", values)
  File "/usr/local/lib/python3.4/dist-packages/WebIOPi-0.7.1-py3.4-linux-armv7l.egg/webiopi/clients/__init__.py", line 97, in sendRequest
    return self.client.sendRequest(method, self.path + path)
  File "/usr/local/lib/python3.4/dist-packages/WebIOPi-0.7.1-py3.4-linux-armv7l.egg/webiopi/clients/__init__.py", line 77, in sendRequest
    raise Exception("No data received")
Exception: No data received
````

It seems that Macro.call() expects a value to be returned from the remote Pi macros, so I added `return 0` statements at the end of each macro on the remote Pi, but the error message persists.

Any ideas what I could try to fix this issue?

thortex, 2016-12-25:
I know that the problem exists, and it has been neglected for a long time... 

I will start to investigate in a few days.

thortex, 2016-12-27:
I fixed this issue in 9db550e. modifications are listed below:

- Added 'bind' function to client socket side to receive response: https://github.com/thortex/rpi3-webiopi/commit/9db550eca34922171bf30d961173183e5daaeee4#diff-17dc48d05dbd06438c193bd6122d1342R413
As Unix network programming, 'recvfrom'/'recv' functions should be used by binding socket diagram (UDP) connection ('bind' function is required to receive a diagram socket server response for a diagram socket client ). Please refer to "UNIX Network Programming" by Richard Stevens for more details.
- Client/server socket timeout was too short to wait remote response (only one second),  I changed to three seconds: https://github.com/thortex/rpi3-webiopi/commit/9db550eca34922171bf30d961173183e5daaeee4#diff-17dc48d05dbd06438c193bd6122d1342L410
- Added SO_REUSEADDR socket option: https://github.com/thortex/rpi3-webiopi/commit/9db550eca34922171bf30d961173183e5daaeee4#diff-17dc48d05dbd06438c193bd6122d1342R412
- Substituted 'recvfrom' for 'recv' function: https://github.com/thortex/rpi3-webiopi/commit/9db550eca34922171bf30d961173183e5daaeee4#diff-17dc48d05dbd06438c193bd6122d1342L419

sumghai, 2016-12-28:
I'll try it out this weekend, thanks!

sumghai, 2016-12-31:
Okay, so I downloaded the latest version of the repository, rebuilt and reinstalled WebIOPi on both of my Raspberry Pis.

On the first Pi, I decided to try one of the example scripts that you said worked in your setup - I had to change the client IP address and remove the credentials requirement in `webiopi_0.7.1/examples/clients/webiopi-client-gpio.py `, before adding the path of the script to /etc/webiopi/config:

```
#!/usr/bin/python
from webiopi.clients import *
from time import sleep

# Create a WebIOPi client
#client = PiHttpClient("192.168.1.234")
#client = PiHttpClient("192.168.0.4")
#client = PiMixedClient("192.168.1.234")
#client = PiCoapClient("224.0.1.123")
client = PiCoapClient("192.168.1.101")   # The actual address of my target Pi
#client = PiCoapClient("192.168.0.4")
#client = PiMulticastClient()

#client.setCredentials("webiopi", "raspberry")    # No credentials needed

# RPi native GPIO
gpio = NativeGPIO(client)
gpio.setFunction(25, "out")
state = True

while True:
    # toggle digital state
    state = not state
    print ("GPIO port 25 value: %d" % state)
    gpio.digitalWrite(25, state)
    sleep(5)
```

Running WebIOPi in debug mode gives the following error:

```
pi@HSFM-EntD-Bridge-CtrlTestNode:~ $ sudo webiopi -d -c /etc/webiopi/config
2016-12-31 13:19:27 - WebIOPi - INFO - Starting YA-WebIOPi/0.7.1/Python3.4
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.bankCount to REST GET /GPIO/banks
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.digitalCount to REST GET /GPIO/count
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.digitalRead to REST GET /GPIO/%(channel)d/value
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.digitalWrite to REST POST /GPIO/%(channel)d/value/%(value)d
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.get to REST GET /GPIO/%(channel)d/hwpwm/port
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.getFrequency to REST GET /GPIO/%(channel)d/freq
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.getFunctionString to REST GET /GPIO/%(channel)d/function
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.getHWPWMDuty to REST GET /GPIO/%(channel)d/hwpwm/duty
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.getHWPWMclockSource to REST GET /GPIO/%(channel)d/hwpwm/clock
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.getHWPWMfrequency to REST GET /GPIO/%(channel)d/hwpwm/freq
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.getHWPWMmSMode to REST GET /GPIO/%(channel)d/hwpwm/msmode
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.getHWPWMoutput to REST GET /GPIO/%(channel)d/hwpwm/output
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.getHWPWMperiod to REST GET /GPIO/%(channel)d/hwpwm/period
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.getHWPWMpolarity to REST GET /GPIO/%(channel)d/hwpwm/polarity
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.getPulse to REST GET /GPIO/%(channel)d/pulse
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.outputSequence to REST POST /GPIO/%(channel)d/sequence/%(args)s
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.portRead to REST GET /GPIO/*/integer
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.portWrite to REST POST /GPIO/*/integer/%(value)d
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.pulse to REST POST /GPIO/%(channel)d/pulse/
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.pulseAngle to REST POST /GPIO/%(channel)d/pulseAngle/%(value)f
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.pulseFreq to REST POST /GPIO/%(channel)d/pulseFreq/%(value)f
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.pulseRatio to REST POST /GPIO/%(channel)d/pulseRatio/%(value)f
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.setFunctionString to REST POST /GPIO/%(channel)d/function/%(value)s
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.setHWPWMDuty to REST POST /GPIO/%(channel)d/hwpwm/duty/%(duty)d
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.setHWPWMclockSource to REST POST /GPIO/%(channel)d/hwpwm/clock/%(src)s
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.setHWPWMfrequency to REST POST /GPIO/%(channel)d/hwpwm/freq/%(value)f
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.setHWPWMmSMode to REST POST /GPIO/%(channel)d/hwpwm/msmode/%(msmode)d
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.setHWPWMoutput to REST POST /GPIO/%(channel)d/hwpwm/output/%(value)s
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.setHWPWMperiod to REST POST /GPIO/%(channel)d/hwpwm/period/%(period)d
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.setHWPWMpolarity to REST POST /GPIO/%(channel)d/hwpwm/polarity/%(polarity)d
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.setHWPWMport to REST POST /GPIO/%(channel)d/hwpwm/port/%(port)d
2016-12-31 13:19:27 - WebIOPi - DEBUG - Mapping GPIO.setPullUpDnControl to REST POST /GPIO/%(channel)d/pullupdn/%(value)s
[... 17 lines cut ...]
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/WebIOPi-0.7.1-py3.4-linux-armv7l.egg/webiopi/__main__.py", line 75, in <module>
    main(sys.argv)
  File "/usr/local/lib/python3.4/dist-packages/WebIOPi-0.7.1-py3.4-linux-armv7l.egg/webiopi/__main__.py", line 69, in main
    server = Server(port=port, configfile=configfile, scriptfile=scriptfile)
  File "/usr/local/lib/python3.4/dist-packages/WebIOPi-0.7.1-py3.4-linux-armv7l.egg/webiopi/server/__init__.py", line 75, in __init__
[... cut ...]
    gpio.digitalWrite(25, state)
  File "/usr/local/lib/python3.4/dist-packages/WebIOPi-0.7.1-py3.4-linux-armv7l.egg/webiopi/clients/__init__.py", line 127, in digitalWrite
    return int(self.sendRequest("POST", "/%d/value/%d" % (channel, value)))
ValueError: invalid literal for int() with base 10: ''
pi@HSFM-EntD-Bridge-CtrlTestNode:~ $
```

I'm guessing something in `gpio.digitalWrite(25, state)` is expecting an integer as input? I'm not entirely sure what's different between your setup and mine...

thortex, 2016-12-31:
My python version is 2, on the other hand, your python version is 3.
So, there are some syntax differences between them.
Please declare 'state' variable as an integer type:

    state = 1
    while True:
    	# toggle digital state
    	if state != 0:
    	  state = 0
    	else:
    	  state = 1
    	print ("GPIO port 25 value: %d" % state)

sumghai, 2016-12-31:
Ah, okay, I'll try that in a moment, and let you know soon.

sumghai, 2016-12-31:
I updated `webiopi_0.7.1/examples/clients/webiopi-client-gpio.py` as follows:

```
#!/usr/bin/python
from webiopi.clients import *
from time import sleep

# Create a WebIOPi client
#client = PiHttpClient("192.168.1.234")
#client = PiHttpClient("192.168.0.4")
#client = PiMixedClient("192.168.1.234")
#client = PiCoapClient("224.0.1.123")
client = PiCoapClient("192.168.1.101")
#client = PiCoapClient("192.168.0.4")
#client = PiMulticastClient()

#client.setCredentials("webiopi", "raspberry")

# RPi native GPIO
gpio = NativeGPIO(client)
gpio.setFunction(25, "out")
	
state = 1
while True:
	# toggle digital state
	if state != 0:
	  state = 0
	else:
	  state = 1
	
	print ("GPIO port 25 value: %d" % state)
	gpio.digitalWrite(25, state)
	sleep(5)
```

Which results in this error message (and WebIOPi terminating):

```
pi@HSFM-EntD-Bridge-CtrlTestNode:~ $ sudo webiopi -d -c /etc/webiopi/config
2016-12-31 14:15:57 - WebIOPi - INFO - Starting YA-WebIOPi/0.7.1/Python3.4
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.bankCount to REST GET /GPIO/banks
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.digitalCount to REST GET /GPIO/count
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.digitalRead to REST GET /GPIO/%(channel)d/value
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.digitalWrite to REST POST /GPIO/%(channel)d/value/%(value)d
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.get to REST GET /GPIO/%(channel)d/hwpwm/port
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.getFrequency to REST GET /GPIO/%(channel)d/freq
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.getFunctionString to REST GET /GPIO/%(channel)d/function
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.getHWPWMDuty to REST GET /GPIO/%(channel)d/hwpwm/duty
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.getHWPWMclockSource to REST GET /GPIO/%(channel)d/hwpwm/clock
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.getHWPWMfrequency to REST GET /GPIO/%(channel)d/hwpwm/freq
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.getHWPWMmSMode to REST GET /GPIO/%(channel)d/hwpwm/msmode
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.getHWPWMoutput to REST GET /GPIO/%(channel)d/hwpwm/output
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.getHWPWMperiod to REST GET /GPIO/%(channel)d/hwpwm/period
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.getHWPWMpolarity to REST GET /GPIO/%(channel)d/hwpwm/polarity
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.getPulse to REST GET /GPIO/%(channel)d/pulse
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.outputSequence to REST POST /GPIO/%(channel)d/sequence/%(args)s
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.portRead to REST GET /GPIO/*/integer
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.portWrite to REST POST /GPIO/*/integer/%(value)d
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.pulse to REST POST /GPIO/%(channel)d/pulse/
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.pulseAngle to REST POST /GPIO/%(channel)d/pulseAngle/%(value)f
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.pulseFreq to REST POST /GPIO/%(channel)d/pulseFreq/%(value)f
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.pulseRatio to REST POST /GPIO/%(channel)d/pulseRatio/%(value)f
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.setFunctionString to REST POST /GPIO/%(channel)d/function/%(value)s
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.setHWPWMDuty to REST POST /GPIO/%(channel)d/hwpwm/duty/%(duty)d
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.setHWPWMclockSource to REST POST /GPIO/%(channel)d/hwpwm/clock/%(src)s
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.setHWPWMfrequency to REST POST /GPIO/%(channel)d/hwpwm/freq/%(value)f
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.setHWPWMmSMode to REST POST /GPIO/%(channel)d/hwpwm/msmode/%(msmode)d
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.setHWPWMoutput to REST POST /GPIO/%(channel)d/hwpwm/output/%(value)s
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.setHWPWMperiod to REST POST /GPIO/%(channel)d/hwpwm/period/%(period)d
2016-12-31 14:15:57 - WebIOPi - DEBUG - Mapping GPIO.setHWPWMpolarity to REST POST /GPIO/%(channel)d/hwpwm/polarity/%(polarity)d
[... 23 lines cut ...]
    server = Server(port=port, configfile=configfile, scriptfile=scriptfile)
  File "/usr/local/lib/python3.4/dist-packages/WebIOPi-0.7.1-py3.4-linux-armv7l.egg/webiopi/server/__init__.py", line 75, in __init__
    loader.loadScript(name, source, self.restHandler)
  File "/usr/local/lib/python3.4/dist-packages/WebIOPi-0.7.1-py3.4-linux-armv7l.egg/webiopi/utils/loader.py", line 8, in loadScript
    script = imp.load_source(name, source)
  File "/usr/lib/python3.4/imp.py", line 171, in load_source
    module = methods.load()
  File "<frozen importlib._bootstrap>", line 1220, in load
  File "<frozen importlib._bootstrap>", line 1200, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1129, in _exec
  File "<frozen importlib._bootstrap>", line 1471, in exec_module
  File "<frozen importlib._bootstrap>", line 321, in _call_with_frames_removed
  File "/home/pi/rpi3-webiopi/webiopi_0.7.1/examples/clients/webiopi-client-gpio.py", line 29, in <module>
    gpio.digitalWrite(25, state)
  File "/usr/local/lib/python3.4/dist-packages/WebIOPi-0.7.1-py3.4-linux-armv7l.egg/webiopi/clients/__init__.py", line 127, in digitalWrite
    return int(self.sendRequest("POST", "/%d/value/%d" % (channel, value)))
ValueError: invalid literal for int() with base 10: "bytearray(b'0')"
pi@HSFM-EntD-Bridge-CtrlTestNode:~ $
```

On a pinch, I tried explicitly casting state to int (i.e. `gpio.digitalWrite(25, int(state))`), but that didn't work, either.

I'm not sure if I should downgrade to Python 2, since that *might* break some of my other scripts.

Thoughts?

thortex, 2016-12-31:
I've fixed this issue in https://github.com/thortex/rpi3-webiopi/commit/5aa5ce472701923c6ed01a908ae16293ee058464#diff-00bc97f7efb4c99627371667989f1357R54 

Modifications are listed below:
* Substituted 'bytearray' object decode() function for 'str' object re-interpreted cast.
* Declared "state" variable as an integer type instead of Boolean type in example script(s).

sumghai, 2017-01-01:
No luck on my end I'm afraid - still getting the `ValueError: invalid literal for int() with base 10: "bytearray(b'0')"` error message.

Incidentally, I just noticed that the bug report guideline asks for the output of `webiopi-diag`, so I tried running that command, but I got a `-bash: webiopi-diag: command not found` error. I also later discovered that several webiopi script files were missing, and that attempting to do a clean uninstall/install didn't help.

I'm going to try manually adding the `webiopi-diag` and `webiopi-diag.sh` files into the `/usr/bin` directory on both my Raspberry Pis. If that works, I should be able to get the diagnostics log which might help diagnose my original CoAP issue.

In case I haven't said so earlier, I really appreciate the hard work you've put into maintaining WebIOPi :)

sumghai, 2017-01-01:
I've posted my `webiopi-diag` logs on Google Drive for your reference:

HSFM-EntD-Bridge-CtrlTestNode (192.168.1.100) - [Log](https://drive.google.com/file/d/0B3zePiS4VoUWQjFFakpVNXRvQm8/view?usp=sharing)

HSFM-EntD-Bridge-AlertLights-1 (192.168.1.101) - [Log](https://drive.google.com/file/d/0B3zePiS4VoUWVktVUUpNQWp2YmM/view?usp=sharing)

thortex, 2017-01-01:
For now, please install python2-webiopi package to 192.168.1.100 for installing basic scripts (python3-webiopi package only includes a native c library and python module scripts without htdocs, init, systemd, and etc.).

thortex, 2017-01-01:
There is an important difference between:
> invalid literal for int() with base 10: "bytearray(b'0')"
> invalid literal for int() with base 10: ''`

In case of `bytearray(b'0')`, old WebIOPi or old Yet Another WebIOPi is running (CoAP related bug is not fixed).
In case of `''`, migrating to latest Yet Another WebIOPI is successful, but, receiving CoAP server response may be in failure on 192.168.1.100 (CoAP client). 

It may be useful for you to enabling log output in protocols/coap.py (the following lines):
- https://github.com/thortex/rpi3-webiopi/blob/master/webiopi_0.7.1/python/webiopi/protocols/coap.py#L421
- https://github.com/thortex/rpi3-webiopi/blob/master/webiopi_0.7.1/python/webiopi/protocols/coap.py#L427
- https://github.com/thortex/rpi3-webiopi/blob/master/webiopi_0.7.1/python/webiopi/protocols/coap.py#L432

Please replace line 427 to the following line:
> print("Received Response:\n%s" % response)
If CoAP server response is received, response message is displayed out.

thortex, 2017-01-01:
OTOH, The possible causes are as follows:

1. Multicast UDP packet is blocked by a wireless router connected by RPi3 (simple multicast UDP client/server program written in another language maybe useful to check configurations). 
2. Wi-Fi interface of RPi3 can not send multicast UDP packet as default (`ip addr | grep wlan0
`).
3. Configuring Wi-Fi interface into promiscuous mode to receive all multicast packets (`sudo ifconfig wlan0 promisc && ifconfig wlan0 | grep PROMISC`).

closed 2017-01-01 by thortex

thortex, 2017-01-09:
I've committed a simple multicast/unicast UDP socket client to check WebIOPI CoAP multicast/unicast socket server on Raspberry Pi in https://github.com/thortex/rpi3-webiopi/commit/d7b696c5185e8708d11137672dc00ab44bb2fcd8. 

Compilation and execution instructions are listed below:
```
    # build C source files on another platform (e.g. Mac OS X, x86/x64 Linux, BSD, et al.).
    $ gcc -o coap-multicast-udp coap-multicast-udp.c
    $ ./coap-multicast-udp 
    Sent a message.
    Received response message from server:
    [60 44 00 00 b4 47 50 49 4f 02 32 35 08 66 75 6e 63 74 69 6f 6e 03 6f 75 74 ff 4f 55 54 ]
    # in case of unicast, specify RPi's IP address (my RPi has the IP address: 192.168.0.4 on wlan0).
    $ gcc -o coap-unicast-udp coap-unicast-udp.c
    $ ./coap-unicast-udp 192.168.0.4
    Sent a message.
    Received response message from server:
    [60 44 00 00 b4 47 50 49 4f 02 32 35 08 66 75 6e 63 74 69 6f 6e 03 6f 75 74 ff 4f 55 54 ]
```

Assume that WebIOPi is invoked by the following command: 
```
    $ sudo python3 -d -m webiopi -d -c /etc/webiopi/config
```

Then, log messages are shown such as the followings:
```
    ...
    2017-01-09 10:25:04 - WebIOPi - INFO - HTTP Server binded on http://192.168.0.4:8000/
    2017-01-09 10:25:04 - WebIOPi - INFO - CoAP Server binded on coap://224.0.1.123:5683/ (MULTICAST)
    2017-01-09 10:25:04 - WebIOPi - INFO - CoAP Server binded on coap://192.168.0.4:5683/
    ... 
    2017-01-09 22:11:29 - CoAP - DEBUG - "POST /GPIO/25/function/out CoAP/1.0" - 2.04 Changed (Client: 192.168.0.4)
    2017-01-09 22:11:29 - CoAP - DEBUG - Received Request:
    Version: 1
    Type: CON
    Code: POST
    Id: 0
    Token: None
    Uri-Path: /GPIO/25/value/0
    Content-Format: text/plain
    Payload: 
    
    2017-01-09 22:11:29 - CoAP - DEBUG - Sending Response:
    Version: 1
    Type: ACK
    Code: 2.04 Changed
    Id: 0
    Token: None
    Uri-Path: /GPIO/25/value/0
    Content-Format: text/plain
    Payload: 0
    
    2017-01-09 22:11:29 - CoAP - DEBUG - "POST /GPIO/25/value/0 CoAP/1.0" - 2.04 Changed (Client: 192.168.0.4)
```

My USB Wi-Fi Dongle of RPi is Buffalo WLI-UC-GNM (Ralink 802.11n chip), and a remote computer is Mac OS X and Wi-Fi.

sumghai, 2017-01-11:
@thortex That looks useful, thanks.

I recently performed a clean installation of Raspbian + Yet Another Webiopi+ - I got some response from CoAP communications, but the behaviour appears to be inconsistent.

Will report back once I've tested things more thoroughly.