thortex/rpi3-webiopi #44: CoAP multicast over wireless network interferes with def loop()

sumghai, 2016-12-24:
Tangentially related to #43.

Enabling CoAP and multicasting seems to cause the loop function in the Python script to only run once and stop. [As Eric (trouch) noted](https://groups.google.com/forum/#!searchin/webiopi/disable$20coap|sort:relevance/webiopi/DochxXTsJj4/Yf_8Cg6ie5EJ), the problem is particularly prevalent on RPis communicating over wireless networks.

[Some users have speculated](https://groups.google.com/forum/#!topic/webiopi/ZVX_mO0LIec) that around the time trouch left WebIOPi for myDevices, the last release of WebIOPi broke CoAP compatibility for RPi 2 & 3.

I know it's probably a big ask of you to look into bringing back CoAP for rpi3-webiopi, but if there is any further information that I can provide to help you diagnose this issue, I'd be happy to help.

thortex, 2016-12-25:
Thank you for your reporting this issue. 
I know that the problem exists, and it has been neglected for a long time... 
I will start to investigate in a few days.

thortex, 2016-12-26:
In my environment, the problem was solved. please try it.

sumghai, 2016-12-31:
After updating, rebuilding and reinstalling WebIOPi with the changes from 9db550e and re-enabling both CoAP and multicast in `/etc/webiopi/config`, the `loop()` function in all scripts work properly on startup.

You can consider #44 to be fixed - I'm still investigating #43.

closed 2016-12-31 by sumghai