thortex/rpi3-webiopi #46: [Scripts] in Config not working if Webiopi started with systemctl

bill-orange, 2017-03-09:
I am using a script to make a GPIO out a momentary contact, cleverly named 'spring_loaded.py'.  I notice that when I start Webiopi with systemctl as shown in the install instructions, loop() in my script does not work.  It is properly invoked in the config.  Setup() and destroy() work but not loop().  If I invoke Webiopi with a shell command line loop() runs fine.

Is this a bug, a limitation of systemctl, am I doing something wrong or am I leaving something out?

I suppose I can invoke it with the command line but this seems odd.

Any advice?

thortex, 2017-03-09:
There are many differences between a command line launch application and a systemctl daemon.
As a work-around, If insert the following python program lines in the top of file: 'spring_loaded.py', does loop() function work properly?

     import time
     time.sleep(30)

bill-orange, 2017-03-10:
Yes! That does the trick, but how?

closed 2017-03-11 by bill-orange

thortex, 2017-03-11:
Hmm.... Maybe something is wrong in 'After' dependency of WebIOPi unit in systemd.unit file: "webiopi.service".
If waiting for network configuration (bring up) is added to Setup() function, loop() function may work fine.