thortex/rpi3-webiopi #48: Loops Crash after a boot / works fine from service manualy started...

Klhnikov, 2017-06-03:
Hey, first a big thank you for your work ! I updated from std Webiopi to your fork and I am realy pleased to see my small robot finaly moving the good way (PWM's looks a lot stronger in that version ...) !

(french speaking here sorry for eventual mistakes...)

Anyway, I went to this Webiopi+ for a reason, my Loop() used to crash with the std webiopi after boot only (everything worked fine except the loop, wich start, make one turn and crash). 
And I have exactely the same behaviour with webiopi+...

After the bug occur, after booting, it is impossible (or veeeery long) to close the webiopi service manualy ("sudo service webiopi stop"), have to "sudo reboot -f" to force it after commenting the whole loop to have a clean (almost) working interface...

An empty loop "def loop(): webiopi.sleep(1)" behave the same...

It may be relevant to say that I use a wifi Dongle on my Pi3 to have : wlan0 : AccessPoint and wlan1 : Internet connexion, my Googling told me it was maybe the cause, but can't figure it out myself, for now...

(ho, not relevant I think but..., the purpose of the loop is to continuously retrieve distance sensor's datas to avoid every wall on my robot's way and start working on an autonomous program... stuck from here...)

Klhnikov, 2017-06-04:
Ok just a message to signal that I re installed all of my environnement from scratch and still had the issue...

Managed to disable COAP and Multicast and everything work fine now...

Should my Robot have troubles without COAP running ? (I use it via accesspoint on the Robot himself most of the time but noticed that controling over local network is tricky... I gess COAP could help ...)

thortex, 2017-06-21:
Hi, Klhnikov.
The original WebIOPi and Yet Another WebIOPi+ still have trouble with COAP running.
I can't fix the issue for a while.

Could you tell me Maker, Product Name, and Product Number of your dongle (or paste the result of "lsusb -v" command output)?

Klhnikov, 2017-06-21:
Hi Thortex !, thanks for answer
No problem, you can find all of my project here (code and hardware) : 
[PiRobot](https://github.com/Klhnikov/PiRobot) (french speaking...)

The USB Wifi dongle I use is a : 
TP-Link TL-WN725N  Nano Adaptateur USB Wi-Fi N 150 Mbps 
[Amazon-link](https://www.amazon.fr/gp/product/B008IFXQFU/ref=oh_aui_detailpage_o07_s00?ie=UTF8&th=1)

thortex, 2017-07-03:
Hi, Klhnikov.
I checked your project PiRobot, and that was great!
I think the python/javascript source codes which you wrote may have no problem with executing WebIOPI.

I will investigate WebIOPi's source codes later.

Klhnikov, 2017-07-04:
Hi ! Yes the robot is awsome !
Im now limited by the rPi spec...

So i'm not really working on it anymore, but its helpfull to watch my sick dog on cam, I use or now, sadly...

Anyway don't mind to much ! But the bug still run ...;)

MichaIng, 2021-04-28:
Same underlying issue like:
- #52
- #53

Sorry for resurrecting all those issues, I hope there is still someone motivated like me to further debug and hopefully fix it 😄.