thortex/rpi3-webiopi #49: Laptop with touch screen and Chrome not working

hmlinaric, 2017-06-19:
Bug description: I have laptop with touch screen, and when i open Chrome browser, buttons (on mouse click) are not sanding any events (not calling macro).

Found the problem and solution within /htdocs/webiopi.js. 
...
...
**//DETECTING IF DEVICE IS TOUCH DEVICE. IF TRUE DISABLE MOUSE - PROBLEM!!!** 
var **isTouchDevice** = "ontouchstart" in document.documentElement ? true : false;
var BUTTON_DOWN   = isTouchDevice ? "touchstart" : "mousedown";
var BUTTON_UP     = isTouchDevice ? "touchend"   : "mouseup";
...
...
...
WebIOPi.prototype.createButton = function (id, label, callback, callbackUp) {
	var button = $('<button type="button" class="Default">');
	button.attr("id", id);
	button.text(label);
	if (callback != undefined) {
		**button.bind(BUTTON_DOWN, callback);**
	}
	if (callbackUp != undefined) {
		**button.bind(BUTTON_UP, callbackUp);**
	}
	return button;
}

In my case my laptop is recognize as  TouchDevice, so mouseup and mousedown event is not bind to button. 
Need to be fixed in future versions.

thortex, 2017-06-21:
Hi, hmlinaric.
Thanks for your bug reporting.
 I will modify source codes, and release it in next milestone; 0.71+deb06.

thortex, 2017-10-01:
Hi, hmlinaric.
Tablet PC (e.g., Surface, Chromebook, et al.) generally has multiple input devices (such a mouse or a multi-touch/-point screen), so both mouse and touch event handlers should be registered in JavaScript.

As as easy way (a work-around), I committed a solution in:
https://github.com/thortex/rpi3-webiopi/commit/285e5268987e9cbb1629251a4802bb8b95b871bd

I wish you would.