thortex/rpi3-webiopi #52: systemctl stop fails on non-debugging mode

thortex, 2018-02-14:
$ sudo systemctl stop webiopi
     $ # takes a while...
     $ sudo systemctl status webiopi
       systemd[1]: webiopi.service: State 'stop-sigterm' timed out. Killing.
       systemd[1]: webiopi.service: Killing Process 12035 (python3) with signal SIGKILL.
       systemd[1]: webiopi.service: Main process exited, code=killed, status=9/KILL.
       systemd[1]: webiopi.service: Unit entered failed state.
       systemd[1]: webiopi.service: Failed with result 'timeout'.

MichaIng, 2021-04-28:
We face the same. I tried it on different Debian Stretch, Buster and Bullseye systems with different Python 2 and 3 versions, with and without debug flag `-d`, SIGTERM leads to infinite 100% CPU usage without ever stopping. systemd SIGKILLs the process after 90 seconds, manually this works as well of course.

`python3 -m webiopi -d` is the easiest way to test this without any systemd and config impact.

Important to note that we use [this](https://github.com/Freenove/WebIOPi) fork, but I also saw it on the original (unpatched) v0.7.1.

I tried `strace python3 -m webiopi -d`, but SIGTERM works there without any additional output. It seems `strace` itself kills the process successfully or probably invokes it in a way there the issue does not appear.

Not sure whether this fork is actively used/developed, but I'm open to help debug and test the underlying issue, if someone with more insights and Python knowledge can point me into the right direction or suggest debug steps I can do.

MichaIng, 2021-04-28:
The HTTP server is the culprit, or how it is handled/threaded. Disabling it, respectively adding `http_enabled = False` in `webiopi/server/__init__.py` solves the issue.

`self.http_server.stop()` is definitely called and I see added debug output after that call.

There `self.server_close()` in the `HTTPServer` as well is called. Checking the docs about the `http.server` class: https://docs.python.org/3/library/http.server.html
- It is setup somehow different, but my Python knowledge is not enough to verify whether this is correct or now.
- [`self.serve_forever()`](https://docs.python.org/3/library/socketserver.html#socketserver.BaseServer.serve_forever) is called as part of the socketserver parent class, but an added debug print after that call is never done.
- [`self.server_close()`](https://docs.python.org/3/library/socketserver.html#socketserver.BaseServer.server_close) is called, an added debug print before and after is shown, but it seems to not have the intended effect (although `ss -tulp` doesn't show the listening port anymore).

### Solution
- [`self.shutdown()`](https://docs.python.org/3/library/socketserver.html#socketserver.BaseServer.shutdown) was missing. Adding it before or after `self.server_close()` successfully stops the webserver and with it the whole WebIOPi process.
- Same on Python 2 btw: https://docs.python.org/2.7/library/socketserver.html#SocketServer.BaseServer.shutdown
- Not sure whether "cleanup" should be done before or after shutting it down, afterwards sounds more reasonable to me, but should be easy to find out.